WifWaf/MH-Z19 #2: High values after power reset

maxbanton, 2019-04-07:
Hi! Thanks for the useful library!
I found a small problem: after a reboot on power, the sensor returns falsely large values, which breaks the display of CO2-level charts.
Here is a serial monitor dump after such reboot:
```CO2 = 60548 ppm
CO2 = 63882 ppm
CO2 = 7011 ppm
CO2 = 4013 ppm
CO2 = 1017 ppm
CO2 = 1016 ppm
CO2 = 1029 ppm
CO2 = 1042 ppm
```
It would be great to add a filter of values to the library if it's possible.

Kind regards, Max.

WifWaf, 2019-04-07:
No problem!

Yes, this is due to the sensor booting up - it outputs a lot of jargon (mostly) too, though it's easier to filter as the CRC fails.

When you request a response during boot it "correctly" returns a value with the CRC valid. However, there is a way to identify it by comparing limited/unlimited or an abnormally large CO2 value (outside of the int range which is used to return the reading).

I've made a crude filter, but I'll get back to you with a version you can test, as I think unexpected power loss and a smooth recovery would be helpful. It will require you to check the error code to see if the filter was triggered so you can ignore the incoming reading - I'll upload an example.

WifWaf, 2019-04-07:
Changes been commit with 8455d3cbb435951e3020e13d224e6fb0d8a61545

Could you give the FilterUsage.ino example a go and see if the same usage works with your code?

Let me know...

maxbanton, 2019-04-08:
Hi! Thanks, I will try.

maxbanton, 2019-04-10:
Hi! I checked the filter - everything works great. Thank you!

closed 2019-04-10 by maxbanton

WifWaf, 2019-04-10:
That's great,

FYI - I'm planing to have the values set to 0 rather than the need to check the errorCodes (bit confusing perhaps?).A statement like If(CO2Reading) can replace the multiple lines (though the value filtered would not be seen).  If there's any issue with this let me know!