WifWaf/MH-Z19 #3: request overlapping

artin-phares, 2019-04-11:
hi, i'm receiving `RESULT_ERR_MATCH` once at startup when using this code: 

```cpp
void setup()
{
    Serial.begin(9600);
    mhz_serial.begin(MHZ_BAUDRATE);

    mhz.printCommunication(false, true);
    mhz.begin(mhz_serial);
    mhz.autoCalibration(false);
}

void loop()
{
    if (millis() - getDataTimer >= 5000)
    {
        int CO2 = mhz.getCO2(false, true);
        Serial.println(CO2); // <-- zero at first (mhz.errorCode == 3 RESULT_ERR_MATCH)

        getDataTimer = millis();
    }
}
```

debug log:
```
---
request:  0xFF 0x01 0x85 0x00 0x00 0x00 0x00 0x00 0x7A  
response: 0xFF 0x85 0x08 0xE5 0x03 0x9E 0x00 0x00 0xED  
---
request:  0xFF 0x01 0xA2 0x00 0x00 0x00 0x00 0x00 0x5D  
response: 0xFF 0xA2 0x08 0xE5 0x03 0x9E 0x00 0x00 0xD0  
---
request:  0xFF 0x01 0x79 0x00 0x00 0x00 0x00 0x00 0x86  
response: 0xFF 0x79 0x01 0x00 0x00 0x00 0x00 0x00 0x86  
---
request:  0xFF 0x01 0x79 0x00 0x00 0x00 0x00 0x00 0x86  
---
request:  0xFF 0x01 0x86 0x00 0x00 0x00 0x00 0x00 0x79  
response: 0xFF 0x79 0x01 0x00 0x00 0x00 0x00 0x00 0x86 ERROR Code: 3
---
request:  0xFF 0x01 0x86 0x00 0x00 0x00 0x00 0x00 0x79  
response: 0xFF 0x86 0x03 0x9F 0x44 0x00 0x00 0x00 0x94  
927
---
```

not sure why 0x79 (ABC turn off) sent twice.  
but first 0x86 (CO2 limited) requested before response for second 0x79 is received, so first 0x86 gets failed.

simillar happens with `examples/BasicUsage`, where co2 and temperature requests overlapping, resulting in failed response for both. fixed by requesting only one at a time.

WifWaf, 2019-04-11:
Hey, 

Yes, It seems to have been sent twice, which has knocked it out of sync. It's not showing up on my Uno - which controller are you using? Are you using the commits from a few hours back or one of the releases?

artin-phares, 2019-04-11:
yeah, i've been using yesterday patch.  
tried latest release (1.4.0) and current patch (7650010) - everything works fine.

probably should have used latest release in the first place. thx.

closed 2019-04-11 by artin-phares

WifWaf, 2019-04-11:
No worries. There's been lots of changes recently (original code was a bit unreliable). Hopefully the next release (over the next few days) is that last for some time.

Glad it's all working for you!