WifWaf/MH-Z19 #4: Consider changing the license

kankaristo, 2019-04-26:
The current license is GPL, which means that you cannot use this library in a project without releasing all of that project's source code under GPL as well.

I don't know if using GPL is a fully thought out choice or not, but it severely restricts where the library can be used.

Please consider releasing changing the license to LGPL (allows linking inside of a program), or to MIT (even better, basically allows all use in a propriety program).

The other 2 libraries I've found for using MH-Z19B are MIT licensed, but one of them only supports PWM (no serial support) and the other one isn't available in Arduino IDE's library manager:
https://github.com/tobiasschuerg/MH-Z-CO2-Sensors
https://github.com/crisap94/MHZ19

Changing the license at this point should be fairly simple, but after you've accepted contributions from other people, they need to agree to the license change as well, as far as I've understood.

The readme says that this library is derived/inspired by another library, which is licensed under GPLv3, so the distinction between derived and inspired is probably important if the license would be changed.

WifWaf, 2019-04-30:
Yes I agree, it's something I've wanted to change as I'm not a fan of the "engulfing" affect the license asserts.

The dilemma is that this library pulled on a few ideas as a started point from a GPL3 library. To my knowledge, the current code in release contains no _direct_ lines of code associated with a GPL license , however there a few similar ideas carried over with some similar nomenclature; I believe the error code usage/formatting.

So to answer your question - I would probably need to contact the owner of the library from which my code was inspired, and ask if he would be willing to release it to me under the MIT/Apache 2.0 licence.  I'll see if this is possible, but obviously no promises...

I will keep this open for now

WifWaf, 2019-05-01:
I've been through the code and believe inspired is the correct description as no code remains (re-written without reference) and it is vastly different overall.

I've updated the license with some changes that were already to commit. However, as a compromise for now, LGPLv3 will be used.

I will create a release over the next week which include the license change.

closed 2019-05-01 by WifWaf