WifWaf/MH-Z19 #5: How to calibrate the sensor properly?

Deliaz, 2019-05-28:
Hi, thank you for this great library.

I've checked and run the calibration example, but it didn't help me. 
I was outside and run sketch couple times without any changes in code (except rx/tx). But mh-z19 still returns me values around 720. Did I do it wrong? 
I have another CO2 monitor, which shows 460-470 at the same spot, and it seems to be a fair number. 

So I noticed that my mh-z19 has wrong delta around 200-300 ppm. Is there a way to make a safe shift for that delta?

WifWaf, 2019-05-30:
Hi Deliaz,

That does sound odd. I tested the current calibration sketch with an UNO and all seems to be OK.

The sketch will take a single reading for calibration when the it's first run or the device reset. So it's usually worth leaving your device outside for 5 minutes before making a restart. The sketch should not restart before new code is uploaded to avoid recalibration (I should probably point this out in the example).  You can also pull the HD pin low instead, by connecting it to ground for at least 7 seconds, which is less awkward.

If you are still having difficulty and if not already, try autoCalibration(true). It uses the lowest CO2 value observed in the previous past 24 hour period as 400ppm and the sensor will recalibrate it's self each day. It usually takes a week or-so to stabilise (though it's not always useful if your environment does not fall to 400ppm - mostly indoors).

I'm not sure if it can be adjusted easily (I think you would need to work with the raw value which is still a bit of mystery). If you were to deduct the difference from a reading, the CO2 value would be inaccurate as the algorithms sensor raw value to CO2 ppm relationship is non-linear.

If you are still having no luck, let me know which code version, sensor and device your using and I'll try and spot an issue.

Deliaz, 2019-05-30:
@WifWaf Thank you. I will try your suggestions. 

> You can also pull the HD pin low instead, by connecting it to ground for at least 7 seconds, which is less awkward.

Does it run calibration on the hardware level, without uploading the sketch?

WifWaf, 2019-05-30:
Yes that's correct, there's no need for any specific sketch.

Deliaz, 2019-06-01:
My appreciation for your help.

I've tried to pull Pin 7 (HD) to the ground and it worked for me. The sensor has reset zero the point, and now values seem to be correct. Thank you. 

But there is another problem: after a few hours of normal work, the sensor stops to respond. Restarting ESP8266 via reset btn doesn't work. It needs to be power reconnected, to make mh-z19 work again. It makes me think the problem is with the sensor itself.

I run ErrorCodes sketch, and it shows something like "timeout reading data" error. Is that possible somehow to debug or explore the problem? 

Thank you. 

[image]

WifWaf, 2019-06-01:
It does read like the sensor is not responding.

If it's running at 3.3v, which is the lower side of the sensors power requirement, you could try 5v in case the power regulation is unstable. 

If you are using yield for WiFi (though I've personally not used the ESP8266), it could possibly be out of sync (though the error message should be different). If you use printCommunication(), and observe the terminal window, you will be able to see what was last received and sent. If the format changes, then it might be synchronisation.

Also, you can try recoveryReset() which tells the sensor to restart, it might help.

WifWaf, 2019-06-01:
Of course trying a different micro controller (I use the ESP32 and UNO) might help see which side the issue is.

WifWaf, 2019-07-07:
Hi Deliaz, I'm going to close this as I believe it's not something directly related to the library. If you are still having issues, I'll see if I can help with in a new issue.

closed 2019-07-07 by WifWaf