WifWaf/MH-Z19 #6: receiving zeros after warmup

artin-phares, 2019-06-03:
hi, sometimes (like 10% chance) i'm receiving zeros from `getCO2` after following init process:

```cpp
void init_mhz() {
  mhz_serial.begin(MHZ_BAUDRATE);

  mhz.begin(mhz_serial);
  mhz.setFilter(true, true);

  mhz.setRange(2000);
  mhz.autoCalibration(true);

  // warming up
  while (true) {
    int co2 = mhz.getCO2(false, true);

    if (mhz.errorCode == RESULT_FILTER) {
      log_ln("sensors: mhz: warming up...", true);
    } else if (mhz.errorCode != RESULT_OK) {
      log_ln("sensors: mhz: failed to read CO2 on warmup.", true);
    } else {
      // warmed up (zeros can slip here)
      break;
    }

    delay(CO2_WARMING_READ_PERIOD);
  }
}
```

i'm using `mhz.errorCode` instead of simplier `co2 == 0` (as in example), because i want to distinguish normal warming (RESULT_FILTER) and failing on warmup (!RESULT_OK).

maybe this is because i'm using `TEMPLIM` instead of `TEMPUNLIM` (what's the difference btw? didn't found any info about that. using `TEMPLIM` because this one specified in manual).

probably `co2 == 0` case should be added to other [filtering cases](https://github.com/WifWaf/MH-Z19/blob/master/src/MHZ19.cpp#L142).

artin-phares, 2019-06-03:
ok, it's even worse.

first co2 reading after startup is 400.
so the sequence is like `400, 0, 0, ..., 0, 0, 620, 622, ...`

init process got fooled with first reading, thinking it's warmed up but it's not.

but with `TEMPUNLIM` (`getCO2(true, true)`) first reading is zero.  
sequence is `0, 0, ..., 0, 0, 620, 622, ...`

unfortunately first reading is still not filtered, second and others until warmed up are filtered.

so im' currently using this workaround:

```cpp
void init_mhz() {
  mhz_serial.begin(MHZ_BAUDRATE);

  mhz.begin(mhz_serial);
  mhz.setFilter(true, true);

  mhz.setRange(2000);
  mhz.autoCalibration(true);

  // warming up
  while (true) {
    int co2 = mhz.getCO2(true, true);

    if (mhz.errorCode == RESULT_FILTER) {
      log_ln("sensors: mhz: warming up...");
    } else if (mhz.errorCode != RESULT_OK) {
      log_ln("sensors: mhz: failed to read CO2 on warmup.", true);
    } else if (co2 == 0) {
      // TODO: remove when zero co2 is filtered
      // https://github.com/WifWaf/MH-Z19/issues/6
      log_ln("sensors: mhz: received zero, continuing warmup...", true);
    } else {
      // warmed up
      break;
    }

    delay(CO2_WARMING_READ_PERIOD);
  }
}
```

WifWaf, 2019-06-06:
You can use setFilter(true, false) to return the offending values, but not set them to 0 - is this what you are looking for? You can then check the RESULT_FILTER code, and choose to ignore the value or store for verification. 

The setFilter(true, true) will set the CO2 values to 0 as you've shown. It allows a simple truthy/falsey check to be used.

The difference between TEMPLIM and TEMPULIM is that the first returns CO2 as per the datasheet command. That is, if your CO2 ppm drops below 400ppm, or exceeds your range, the values are capped, and will not show below/above these thresholds. TEMPULIM simply returns the CO2 ppm regardless is not capped, this command is not found in the datasheets - I should probably rename the enumerations as they are outdated.

I'm not sure why the TEMPLIM returns the 400ppm for the initial value, I'll have to test this... the filter was pretty awkward to setup.

WifWaf, 2020-12-21:
Should be corrected now.

closed 2020-12-21 by WifWaf