WifWaf/MH-Z19 #8: SoftwareSerial not working

haldi4803, 2019-11-05:
`error: no matching function for call to 'SoftwareSerial::SoftwareSerial(int, int)'

 SoftwareSerial mySerial(RX_PIN, TX_PIN);                   // Uno example`

With ESPSoftwareSerial 5.3.4 from: https://github.com/plerup/espsoftwareserial/
Which version do you recommend?

WifWaf, 2019-11-08:
Hi,

Yes, this line only demonstrates the SoftwareSerial functionality for an UNO. The library behind it will be different.

I haven't used the ESP8266 before (looking at the library you've provided), however, you should be able to create an instance of the library and pass it to my library. I'm not sure about versions/library choice.

Have a look at the [example provided](https://github.com/plerup/espsoftwareserial/blob/master/examples/onewiretest/onewiretest.ino)

So.. something like (I can't test this, unfortunately).

> MHZ19 myMHZ19; 
> SoftwareSerial swSer
> 
> void setup() 
> {
>    delay(2000);
> 
>    Serial.begin(115200);
>    Serial.println("\nOne Wire Half Duplex Serial Tester");
> 
>    swSer.begin(115200, 14, 14, SWSERIAL_8N1, false, 256);
>    swSer.enableIntTx(true);
> 
>    myMHZ19.begin(swSer).
> }

[Have a look at the header ](https://github.com/plerup/espsoftwareserial/blob/master/src/SoftwareSerial.h)file they are using for the software serial .begin() arguments.  Line 53.

positron96, 2019-11-23:
By the way, the built-in SoftwareSerial library in ESP8266 core works fine with MH-Z19 and this library. It has same interface as AVR library (accepts 2 pins in constructor).

WifWaf, 2019-12-18:
Closing due to inactivity, feel to open a new issue.

closed 2019-12-18 by WifWaf