WifWaf/MH-Z19 #9: Why have BAUD_RATE in every sketch when it can be incapsulated into library

positron96, 2019-11-16:
Since baud rate of sensor must always be 9600, it could be hidden inside library, e.g. call stream.begin(9600) inside myMHZ19.begin(stream) or add a separate begin method that takes 2 pins and creates a SoftwareSerial object and opens it.

WifWaf, 2019-11-21:
Hi, I haven't worked with the Serial protocols for a while, however, my understanding was that the Stream class is a base class, so cannot invoke a begin() function, as this would be calling itself.

If I were to create a function with pin arguments, then I would be limiting that Stream to a particular board/library, which would have issues.

I believe this was why it's where it is now.

WifWaf, 2019-11-21:
Just to be clear on my interpretation - the Stream class handles the universal Arduino part, the begin function is a device/library dependent configuration.

positron96, 2019-11-23:
Hi, thanks for the response. 
I realized that ``begin(speed)`` is not the part of Stream class when trying to modify your library for my small project, so I ended up with two constructors:
```
	void begin(HardwareSerial &stream, bool setSpeed=true) {
		mySerial = &stream;
		if(setSpeed) stream.begin(9800);
	}
	void begin(SoftwareSerial &stream, bool setSpeed=true) {
		mySerial = &stream;
		if(setSpeed) stream.begin(9800);
	}
```

It's not elegant at all, but usage of the library is clearer.

WifWaf, 2019-11-23:
Hey,  thanks for the suggestion.

Unfortunately, not all Arduino packages include the SoftwareSerial library as default, such as the with the ESP32. I might be a bit confusing to have two different setup methods as I think SoftwareSerial is quite common on the Uno. 

Though I agree, it is not very clear at the moment and something that stood out to me when updating the library a few weeks back...

WifWaf, 2019-12-18:
Closing this for now - I don't believe there is currently a solution. Thanks for prior suggestions,

closed 2019-12-18 by WifWaf