WifWaf/MH-Z19 #10: BasicUsage fails on the Wemos D1 Mini

hixfield, 2019-12-16:
Using :

#define RX_PIN D5                                          // Rx pin which the MHZ19 Tx pin is attached to
#define TX_PIN D6                                          // Tx pin which the MHZ19 Rx pin is attached to
#define BAUDRATE 9600                               // Native to the sensor (do not change)

in the basicusage sketch.

Other libraries are able to fetch the temperature and ppm, this library fails.
I hooked up an oscilloscope and can see that the device indeed replies (TX and RX are both active).

Output of the sketch:

!ERROR: Failed to verify connection(1) to sensor. Failed to stablise
!ERROR: Initial communication errorCode recieved
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17
CO2 (ppm): 0
Temperature (C): -17

hixfield, 2019-12-17:
#11

WifWaf, 2019-12-17:
Hi,

Yes, the datasheets specify a baud-rate of 9600, so I imagine the sensor sequential bytes are arriving too late for the uart configuration. I.e. the interval between available() and read() is too small. 

The timeout would not help here, as the while condition is broken (data was available, just still arriving slow).

I'll have a better look at the pull request tomorrow, however, I can see your thought process.

Cheers,

closed 2019-12-18 by hixfield