WifWaf/MH-Z19 #15: Trouble with HardwareSerial on nodemcu 12

w00dwind, 2020-03-22:
Hello, thank for your library, it's very useful, especially Recovery and Calibrating part.
When i use SoftwareSerial, everything work great, but 3 hours later i don't receive any data from the sensor (i tested it several times). So i read, that using HardwareSerial on ESP gives more stability, and i tried to connect via HardwareSerial, but don't received any data. Maybe something wrong with my sketch, but i tried to minimum change it from original example sketch. 
I just change strings 

> HardwareSerial mySerial(1);    
to
> HardwareSerial Serial2(2);

and also 

> mySerial.begin(BAUDRATE, SERIAL_8N1, RX_PIN, TX_PIN);
to

> Serial2.begin(BAUDRATE, SERIAL_8N1); 

Cause if i try to compile it with original code, i have error:
`var/folders/j2/rv6w5j790dq31zz952rgp0j00000gn/T/arduino_modified_sketch_904602/BasicUsage.ino: In function 'void setup()':
BasicUsage:21:56: error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]
     mySerial.begin(BAUDRATE, SERIAL_8N1, RX_PIN, TX_PIN); // (ESP32 Example) device to MH-Z19 serial start   
                                                        ^
In file included from /Users/ac1d/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/cores/esp8266/Arduino.h:244:0,
                 from /var/folders/j2/rv6w5j790dq31zz952rgp0j00000gn/T/arduino_modified_sketch_904602/BasicUsage.ino:1:
/Users/ac1d/Library/Arduino15/packages/esp8266/hardware/esp8266/2.6.2/cores/esp8266/HardwareSerial.h:87:10: error:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)' [-fpermissive]
     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
          ^
exit status 1
invalid conversion from 'int' to 'SerialMode' [-fpermissive]
`
 Connections:
MHZ19 TX -> GPIO13 (D7) (RXD2)
MHZ19 RX -> GPIO15 (D8) (TXD2)

`#include <Arduino.h>
#include "MHZ19.h"                                        
//#include <SoftwareSerial.h>                                // Remove if using HardwareSerial or Arduino package without SoftwareSerial support
#include "HardwareSerial.h"

//#define RX_PIN 10                                          // Rx pin which the MHZ19 Tx pin is attached to
//#define TX_PIN 11                                          // Tx pin which the MHZ19 Rx pin is attached to
#define BAUDRATE 9600                                      // Device to MH-Z19 Serial baudrate (should not be changed)

MHZ19 myMHZ19;                                             // Constructor for library

//SoftwareSerial mySerial(RX_PIN, TX_PIN);                   // (Uno example) create device to MH-Z19 serial
HardwareSerial Serial2(2);                                // (ESP32 Example) create device to MH-Z19 serial

unsigned long getDataTimer = 0;

void setup()
{
    Serial.begin(9600);                                     // Device to serial monitor feedback
    //Serial.swap(); //GPIO15 (TX) and GPIO13 (RX)
    //Serial.flush(); //clear serial buffer
    //mySerial.begin(BAUDRATE);                               // (Uno example) device to MH-Z19 serial start   
    Serial2.begin(BAUDRATE, SERIAL_8N1);                    // (ESP32 Example) device to MH-Z19 serial start   
    
    
    myMHZ19.begin(Serial2);                                // *Serial(Stream) refence must be passed to library begin(). 
    
    myMHZ19.autoCalibration();                              // Turn auto calibration ON (OFF autoCalibration(false))
}

void loop()
{
    if (millis() - getDataTimer >= 2000)
    {
        int CO2; 

        /* note: getCO2() default is command "CO2 Unlimited". This returns the correct CO2 reading even 
        if below background CO2 levels or above range (useful to validate sensor). You can use the 
        usual documented command with getCO2(false) */

        CO2 = myMHZ19.getCO2();                             // Request CO2 (as ppm)
        
        Serial.print("CO2 (ppm): ");                      
        Serial.println(CO2);                                

        int8_t Temp;
        Temp = myMHZ19.getTemperature();                     // Request Temperature (as Celsius)
        Serial.print("Temperature (C): ");                  
        Serial.println(Temp);                               

        getDataTimer = millis();
    }
}`

WifWaf, 2020-03-23:
Hi, no problem. 

It looks like the error message is due to passing UART pins to begin function [which is expecting a serialmode definition instead](https://github.com/esp8266/Arduino/blob/007e495e0d88eabc6dae31fad311df38f8914607/cores/esp8266/HardwareSerial.h#L92). It seems you can [set the pins with another function](https://github.com/esp8266/Arduino/blob/1127a090adc1a2469068f57d78db79f0c2d636df/cores/esp8266/HardwareSerial.h#L123) in setup the phase. Unfortunately, I haven't used an esp8266 before so I can't test it. 

Have you been able to receive any data from the sensor since it stopped working?  You could try and power on / power off  / use the reset command for the sensor (it puts out data on reboot), and read the serial to see if it is transmitting. For example, if your loop:
 
> while(Serial.available() > 0) {
>     byte  incomingByte = Serial.read();
>     Serial.print("I received: ");
>     Serial.println(incomingByte, DEC);
>   }

WifWaf, 2020-07-05:
Closing this now as it seems resolved.

closed 2020-07-05 by WifWaf

pondinesh006, 2021-07-04:
I am getting the same problem, can you tell me the resolving method to solve this problem and why it is happening

WifWaf, 2021-07-04:
Are using the ESP8266? 

If so it's due to the hardware limitations of the esp8266 rather than this library. My understanding is the ESP8266 has only one free hardware port which is used by Serial.print.

Software Serial should work however, but if it's not, can you post your begin function code.

pondinesh006, 2021-07-05:
/* Comment this out to disable prints and save space */
#define BLYNK_PRINT Serial

// Select your modem:
#define TINY_GSM_MODEM_SIM800
//#define TINY_GSM_MODEM_SIM900
//#define TINY_GSM_MODEM_M590
//#define TINY_GSM_MODEM_A6
//#define TINY_GSM_MODEM_A7
//#define TINY_GSM_MODEM_BG96
//#define TINY_GSM_MODEM_XBEE

// Default heartbeat interval for GSM is 60
// If you want override this value, uncomment and set this option:
//#define BLYNK_HEARTBEAT 30

#include <TinyGPS++.h> //https://github.com/mikalhart/TinyGPSPlus
#include <TinyGsmClient.h> //https://github.com/vshymanskyy/TinyGSM
#include <BlynkSimpleTinyGSM.h> //https://github.com/blynkkk/blynk-library

// You should get Auth Token in the Blynk App.
// Go to the Project Settings (nut icon).
char auth[] = "";

// Your GPRS credentials
// Leave empty, if missing user or pass
char apn[]  = "airtelgprs.com";
char user[] = "";
char pass[] = "";

//sender phone number with country code.
//not gsm module phone number
//const String PHONE = "Enter_Your_Phone_Number";

//GSM Module Settings
//GSM Module RX pin to ESP8266 1 (ESP32 2)
//GSM Module TX pin to ESP8266 3 (ESP32 4)
#define rxPin 3
#define txPin 1
HardwareSerial sim800(1);
TinyGsm modem(sim800);

//GPS Module Settings
//GPS Module RX pin to ESP8266 5(ESP32 17)
//GPS Module TX pin to ESP8266 4(ESP32 16)
#define RXD2 4
#define TXD2 5
HardwareSerial neogps(2);
TinyGPSPlus gps;

WidgetMap myMap(V0);
BlynkTimer timer;
int pointIndex = 1;

void setup() {
   
  //Set Serial monitor baud rate
  Serial.begin(115200);
  Serial.println("esp32 serial initialize");
  delay(10);
  
  //Set GPS module baud rate
  neogps.begin(9600, SERIAL_8N1, RXD2, TXD2); 
  Serial.println("neogps serial initialize");
  delay(10);

  //Set GSM module baud rate
  sim800.begin(9600, SERIAL_8N1, rxPin , txPin);
  Serial.println("SIM800L serial initialize");
  delay(3000); 
   
  // Restart takes quite some time
  // To skip it, call init() instead of restart()
  Serial.println("Initializing modem...");
  modem.restart();

  // Unlock your SIM card with a PIN
  //modem.simUnlock("1234");

  Blynk.begin(auth, modem, apn, user, pass);
  //timer.setInterval(5000L, sendToBlynk);
}

void loop() {
  
  while(neogps.available())
  {
    if (gps.encode(neogps.read()))
    {
      sendToBlynk();
    }
  }

  Blynk.run();
  //timer.run();
  
} //main loop ends

void sendToBlynk()
{

  if (gps.location.isValid() )
  {
    //get latitude and longitude
    float latitude = (gps.location.lat());
    float longitude = (gps.location.lng());
    //get
    float speed = gps.speed.kmph(); 
    //get number of satellites
    float satellites = gps.satellites.value();  
    
    Serial.print("Latitude:  ");
    Serial.println(latitude, 6);
    Serial.print("Longitude: ");
    Serial.println(longitude, 6);
    //Serial.print("Speed: ");
    //Serial.println(speed, 6);    
    
    Blynk.virtualWrite(V1, String(latitude, 6));
    Blynk.virtualWrite(V2, String(longitude, 6));
    myMap.location(pointIndex, latitude, longitude, "GPS_Location");
    
    Blynk.virtualWrite(V3, speed);
    Blynk.virtualWrite(V4, satellites);
    // float bearing = TinyGPSPlus::cardinal(gps.course.value()); // get the direction
    // Blynk.virtualWrite(V5, bearing);
  }
}

pondinesh006, 2021-07-05:
This My code, I am Getting error while compiling it in ESP8266 only, but ESP32 DEV board use to compile  it shows no error.
What I can do for this to compile in ESP8266.

WifWaf, 2021-07-05:
I'm not familiar with the ESP8266, so really can't help much.. however, it has only one free Serial port.

I take it "HardwareSerial neogps(2);" is producing the same errors ? If you can post the error messages, I'll have a look... but I would try using SoftwareSerial instead.

pondinesh006, 2021-07-06:
D:\Dinesh\Works\gps\gps_sim\gps_sim.ino: In function 'void setup()':
gps_sim:47:14: error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]
   47 | #define RXD2 4
      |              ^
      |              |
      |              int
D:\Dinesh\Works\gps\gps_sim\gps_sim.ino:64:34: note: in expansion of macro 'RXD2'
   64 |   neogps.begin(9600, SERIAL_8N1, RXD2, TXD2);
      |                                  ^~~~
In file included from C:\Users\pondi\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h:288,
                 from sketch\gps_sim.ino.cpp:1:
C:\Users\pondi\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/HardwareSerial.h:87:68: note:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)'
   87 |     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
      |                                                         ~~~~~~~~~~~^~~~
gps_sim:39:15: error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]
   39 | #define rxPin 3
      |               ^
      |               |
      |               int
D:\Dinesh\Works\gps\gps_sim\gps_sim.ino:69:34: note: in expansion of macro 'rxPin'
   69 |   sim800.begin(9600, SERIAL_8N1, rxPin , txPin);
      |                                  ^~~~~
In file included from C:\Users\pondi\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/Arduino.h:288,
                 from sketch\gps_sim.ino.cpp:1:
C:\Users\pondi\OneDrive\Documents\ArduinoData\packages\esp8266\hardware\esp8266\3.0.0\cores\esp8266/HardwareSerial.h:87:68: note:   initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode, uint8_t)'
   87 |     void begin(unsigned long baud, SerialConfig config, SerialMode mode, uint8_t tx_pin)
      |                                                         ~~~~~~~~~~~^~~~
exit status 1
invalid conversion from 'int' to 'SerialMode' [-fpermissive]

pondinesh006, 2021-07-06:
This show error like this only.

WifWaf, 2021-07-06:
See here: https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html

The esp8266 Serial port is used by USB serial for Serial. print.  Arduino is not the same on different microcontrollers. Here, the Serial. begin does not accommodate pin Rx and Tx definitions because it's used by Serial. print (USB UART). The only pin argument it takes is for Tx (pin 1 or 2) because the other Rx pin is attached to the flash. 

So it looks like you could turn off Serial. prints over USB and use that hardware serial port, but not both or more.

Update your code here:

HardwareSerial sim800(1);
HardwareSerial neogps(2);

To:

SoftwareSerial sim800;
SoftwareSerial neogps;

Unfortunately, I can't help anymore because it's not an issue with this library and I don't own one to test. It's due to the differences between Arduino firmware. You could try asking here: https://www.reddit.com/r/esp8266/.

pondinesh006, 2021-07-06:
Okk, Thank you..

kuldeepaher01, 2022-11-17:
> See here: https://tttapa.github.io/ESP8266/Chap04%20-%20Microcontroller.html
>
> The esp8266 Serial port is used by USB serial for Serial. print. Arduino is not the same on different microcontrollers. [...]

Even if i use software serial i'm getting same error. I want to use both GSM and GPS module. I also tried doing one hardware and other serial but same error

kuldeepaher01, 2022-11-17:
Also the void begin is asking for software mode.. thats what i think from looking at the errors|
`void begin(unsigned long baud, SerialConfig config, SerialMode mode, uni8_t tx_pin)

error: invalid conversion from 'int' to 'SerialMode' [-fpermissive]

initializing argument 3 of 'void HardwareSerial::begin(long unsigned int, SerialConfig, SerialMode)'
                                              
`
Error takes place while calling this and specifically it is because in my code im calling it like 
`sim800.begin(9600, SERIAL_8N1, rxGSM, txGSM);`

pondinesh006, 2022-11-17:
Thank you for your response

pondinesh006, 2022-12-28:
> yep, just change the 'serial.being( 9600, SERIAL_8N1, 16,17)toSerial.begin(9600,
> SERIAL_8N1);`
> On the NODEMCU these are connected to TX and RX (gpio1 and 3)

Thanks, I'll check them out.

And let you know.