WifWaf/MH-Z19 #16: !Error: Timed out waiting for response

EG3007, 2020-04-02:
Hi,
I am trying to use your library. Due to every connection seems okey i am not getting and data from my MHZ-19 sensor.

I constantly get the following error;

 CO2 (ppm): 0
 !Error: Timed out waiting for response
 Temperature (C): -17

Thank you.

WifWaf, 2020-04-05:
Hi, 

It's very hard to know what is causing this.

Which microcontroller are you using? Also, are you using hardware/software serial?  Do you have other devices in your setup apart from the controller and sensor?

Cheers

LazyNecromancer, 2020-07-05:
I'm having the same Issue, not sure if I'm doing something wrong or my sensor/board is just ded. I'm using your "basic usage" example code verbatim with software serial. I'm using an Arduino Nano knock off, with the rx/tx pins heading to D10/D11, the MH-z19 powered from an external 5V source (a USB battery) to make sure it gets enough juice, and hte Arduino itself running serial to my PC via the natrive serial port.

[image]

WifWaf, 2020-07-05:
Yes, the sensor is not being found.

Do you get the same result over hardware serial? And there any other devices connected to the serial bus?

Also, just to cover all bases, are the rx/tx wires are crossed (I.e rx -> tx, tx -> rx)?

LazyNecromancer, 2020-07-07:
[image]
Well, I tried both Hardware and Software Serial, and reversing the tx/rx pins, no other device other than the sensor and the COM port ot PC for the readout. I tried the Analogue code too, I get *some* results from that, but they're nonsense (unless the reason I can't get this working is that my brain is pickled by way too much C02).

I'm going to assume my sensor is just dead.

WifWaf, 2020-07-09:
Was the analog sketch reading from the sensor analog out pin? If so, it does look like nothing is happening on the sensor side. 

I guess the only other thing is to consider power issues/cables...

LazyNecromancer, 2020-07-12:
I got it working! Unbelievably, it looks like the *breadboard* was the culprit, swapping that out worked. I've still got some work to do on calibration, but it shows higher values inside than outside and higher still when I breathe on it, so it's working to some degree at least.

WifWaf, 2020-07-13:
Great stuff - glad you got past the problem. Yep, calibration can be a bit awkward - there should be some notes in the calibration sketch.

closed 2020-11-20 by WifWaf

epvasquez, 2021-06-23:
hello, I hav the same problem, but with 2 MH-Z19C sensor ... I dont know why, I've tried con the Arduino mega, on the Uno and on an esp8266, alll with the same conection problem.. 

maybe its a sensor problem......

WifWaf, 2021-06-24:
The MH-Z9C shouldn't have additional requirements to set up.

Have you tried crossing Rx / Tx and changing the wires etc?

Could you post your setup code too?

epvasquez, 2021-06-24:
just solved my problem, i don know what happend, but now if got them running on the mega and on the esp8266....  with the basicUsage code but with some minor changes for serial comunication. ( mega pins 18 and 19 and esp 8266 on D7 and D8) 

thank you ! 

if any one has the same problem.. here are the codes. 

#include <Arduino.h>
#include "MHZ19.h"                                        
#include <SoftwareSerial.h>                                // Remove if using HardwareSerial

//#define RX_PIN D5                                          // Rx pin which the MHZ19 Tx pin is attached to
//#define TX_PIN D6                                          // Tx pin which the MHZ19 Rx pin is attached to
#define BAUDRATE 9600                                      // Device to MH-Z19 Serial baudrate (should not be changed)

MHZ19 myMHZ19;                                             // Constructor for library
SoftwareSerial s(13,15);                   // (esp8266) create device to MH-Z19 serial

unsigned long getDataTimer = 0;

void setup()
{
    Serial.begin(9600);                                     // Device to serial monitor feedback

    s.begin(BAUDRATE);                               // (Uno example) device to MH-Z19 serial start   
    myMHZ19.begin(s);                                // *Serial(Stream) refence must be passed to library begin(). 

    myMHZ19.autoCalibration();                              // Turn auto calibration ON (OFF autoCalibration(false))
}

void loop()
{
    if (millis() - getDataTimer >= 2000)
    {
        int CO2; 

        /* note: getCO2() default is command "CO2 Unlimited". This returns the correct CO2 reading even 
        if below background CO2 levels or above range (useful to validate sensor). You can use the 
        usual documented command with getCO2(false) */

        CO2 = myMHZ19.getCO2();                             // Request CO2 (as ppm)
        
        Serial.print("CO2 (ppm): ");                      
        Serial.println(CO2);                                

        int8_t Temp;
        Temp = myMHZ19.getTemperature();                     // Request Temperature (as Celsius)
        Serial.print("Temperature (C): ");                  
        Serial.println(Temp);                               

        getDataTimer = millis();
    }
}

and the CODE with MEGA

#include <MHZ19.h>
#include <Arduino.h>
                                      
//#include <SoftwareSerial.h>                                // Remove if using HardwareSerial

//#define RX_PIN 10                                          // Rx pin which the MHZ19 Tx pin is attached to
//#define TX_PIN 11                                          // Tx pin which the MHZ19 Rx pin is attached to
#define BAUDRATE 9600                                      // Device to MH-Z19 Serial baudrate (should not be changed)

MHZ19 myMHZ19;                                             // Constructor for library
//SoftwareSerial mySerial(RX_PIN, TX_PIN);                   // (Uno example) create device to MH-Z19 serial

unsigned long getDataTimer = 0;

void setup()
{
    Serial.begin(9600);                                     // Device to serial monitor feedback
    Serial1.begin(BAUDRATE);                               // (Mega example) device to MH-Z19 serial start   
    myMHZ19.begin(Serial1);                                // *Serial(Stream) refence must be passed to library begin(). 

//    myMHZ19.autoCalibration();                              // Turn auto calibration ON (OFF autoCalibration(false))
}

void loop()
{
    if (millis() - getDataTimer >= 2000)
    {
        int CO2; 

        /* note: getCO2() default is command "CO2 Unlimited". This returns the correct CO2 reading even 
        if below background CO2 levels or above range (useful to validate sensor). You can use the 
        usual documented command with getCO2(false) */

        CO2 = myMHZ19.getCO2();                             // Request CO2 (as ppm)
        
        Serial.print("CO2 (ppm): ");                      
        Serial.println(CO2);                                

        int8_t Temp;
        Temp = myMHZ19.getTemperature();                     // Request Temperature (as Celsius)
        Serial.print("Temperature (C): ");                  
        Serial.println(Temp);                               

        getDataTimer = millis();
    }
}

WifWaf, 2021-06-24:
Great stuff! 

Glad you figured it out, and thanks for leaving your solution for others.

fm-sys, 2021-09-05:
> 
> 
> Yes, the sensor is not being found.
> 
> Do you get the same result over hardware serial? And there any other devices connected to the serial bus?
> 
> Also, just to cover all bases, are the rx/tx wires are crossed (I.e rx -> tx, tx -> rx)?

That's it, thanks! 

I had the same issue with MH-Z19C. Reason was that I had connected `rx -> rx; tx -> tx` which cannot work for sure...

WifWaf, 2021-09-05:
It's an easy mistake, even when you know - glad it's working also.

stouph47, 2021-10-16:
Hello,
I have the same problem with Uno and Grove base Shield V2. Connections Rx->Tx and Tx->Rx, i tried several configurations, always this same message:
CO2 (ppm): 0
!Error: Timed out waiting for response
Temperature (C): -17

I use software Serial.
Anyone knows if is it due to Grove Shield?
Thanks

stouph47, 2021-10-16:
I change RX pin by 1 and TX pin by 2 (as on Grove Shield documentation).
Now i have another issue:
this message appear:
[image]

Any idea of the issue?
Thanks for the help

WifWaf, 2021-10-16:
Looking at the shield it seems to break out the pins into headers? It should work fine.

If you have a multimeter, try a continuity test to make sure there's a good connection.

JohMah, 2021-12-07:
I have the same problem. It works fine when the sensor is supplied from USB with 4.6 V. When I connect a 5V booster to get exactly 5V I get these timeout messages.