WifWaf/MH-Z19 #22: Question about the Calibration

DerPicknicker, 2020-11-11:
Hey, 

thank your for your nice Library and your Documentation about it. I bought this Sensor and I want to use it only indoors and my Question is about the Calibration. I thought that I need to turn of the auto-calibration, right? 

But how can I calibrate this Sensor to have stable Values for a long time? 

Is it possible to use your Calibration code with an simple button? I mean : I pressed the button outside the Sensor calibrates itself and then its done. Or needs the Sensor a longer Calibration Time? 

Could you please explain to me if I disable the Auto-Calibration disabled that also the temperature Compensation? 

Thank you very much for answering my Questions..

WifWaf, 2020-11-12:
Hi,

If it's indoors and there are people around, then it's possible CO2 might not fall to background levels, in that case, it's best to turn autocalibration off.  Depending on what version of the sensor you have, see the pdfs in the extra folder and locate the HD pin - if you pull this low (i.e. a button) for 7 seconds, it will force it to take a reading. It assumes this is background CO2 level (i.e. ~400ppm) - you don't need any code for this,

Turning off autocalibration doesn't affect the temperature compensation that I'm aware of.

In terms of energy, yes you can if you aren't using autocalibration - it should work OK. You'll just have to ignore the warm-up readings. You can use the set filter(true) function in setup, which will cause getCO2 to return 0's until it's recovered.

In terms of stability, it mostly comes down to calibrating at the right time - night time is usually the closest to background CO2 levels.

DerPicknicker, 2020-11-12:
Thank you for your quick Response. 

I can now understand the MH-Z19b a lot better. If I disable the auto calibration, and Use your Method (7secs) for the calibration, could you estimate how long the Sensor needs fresh Air to finish the calibration? 

After I setup the Sensor with filter(true) I can switch of the Sensor and restart it 5 mins later. And the Sensor knows when he is warm enough for correct measurements, right? 

I can’t find something about switching the Sensor of, is it possible to turn it on/off with Code? Or do I need to disconnect the Power Source from the MH-z19b? 

Thank you very much for answering my questions.

DerPicknicker, 2020-11-12:
#include <Arduino.h>
#include "MHZ19.h"                                        
#include <SoftwareSerial.h>                                // Remove if using HardwareSerial or Arduino package without SoftwareSerial support

#define RX_PIN 3                                          // Rx pin which the MHZ19 Tx pin is attached to
#define TX_PIN 4                                          // Tx pin which the MHZ19 Rx pin is attached to
#define BAUDRATE 9600                                      // Device to MH-Z19 Serial baudrate (should not be changed)

MHZ19 myMHZ19;                                             // Constructor for library

SoftwareSerial mySerial(RX_PIN, TX_PIN);                   // (Uno example) create device to MH-Z19 serial
//HardwareSerial mySerial(1);                              // (ESP32 Example) create device to MH-Z19 serial

unsigned long getDataTimer = 0;

void setup()
{
    Serial.begin(9600);                                     // Device to serial monitor feedback

    mySerial.begin(BAUDRATE);                               // (Uno example) device to MH-Z19 serial start   
    //mySerial.begin(BAUDRATE, SERIAL_8N1, RX_PIN, TX_PIN); // (ESP32 Example) device to MH-Z19 serial start   
    myMHZ19.begin(mySerial);                                // *Serial(Stream) refence must be passed to library begin(). 
    delay(10);
    
    //myMHZ19.setFilter(true, true);                          // Default Filter-Mode, if no Correct Values are present the Sensor returns "0"
    myMHZ19.autoCalibration(false);                         // Turn auto calibration ON (OFF autoCalibration(false))
    //myMHZ19.setSpan(2000); 
    Serial.print("Range: ");
    Serial.println(myMHZ19.getRange());   
    Serial.print("Background CO2: ");
    Serial.println(myMHZ19.getBackgroundCO2());
    Serial.print("Temperature Cal: ");
    Serial.println(myMHZ19.getTempAdjustment());
    Serial.print("ABC Status: "); myMHZ19.getABC() ? Serial.println("ON") :  Serial.println("OFF");
}

void loop()
{
    if (millis() - getDataTimer >= 2000)
    {
        int CO2; 

        /* note: getCO2() default is command "CO2 Unlimited". This returns the correct CO2 reading even 
        if below background CO2 levels or above range (useful to validate sensor). You can use the 
        usual documented command with getCO2(false) */

        CO2 = myMHZ19.getCO2();                             // Request CO2 (as ppm)
        
        Serial.print("CO2: ");                      
        Serial.print(CO2);                                
        Serial.print(",");
        int8_t Temp;
        Temp = myMHZ19.getTemperature();                     // Request Temperature (as Celsius)
        Serial.print(" Temperature: ");                  
        Serial.println(Temp);                               

        getDataTimer = millis();
    }
}

WifWaf, 2020-11-14:
Hi, just working through your questions in sequence.

It will take a reading instantly, so wait for the sensor to have warmed up ( ~30 seconds since it's turned one), then just pull the HD pin low for 7 seconds - it's done straight away.

The library knows which readings to ignore when the sensor is warming up - so instead of returning an erroneous value, it will return a  0. It means you can write something like if(lib.getCO2){ code for valid readings }. It will work for as long as it's included (it's not a native sensor feature).

It can't be turned off by code that I know of - if you want to automate it, I would go with a[ lower current dc solid state relay](https://www.digikey.co.uk/products/en/relays/solid-state-relays/183?k=&pkeyword=&sv=0&pv127=45918&pv127=44509&pv127=44768&pv127=45111&pv127=45432&pv127=45556&pv127=49035&pv127=47593&pv127=48020&pv127=51272&pv127=49671&pv127=50598&pv127=54330&pv127=56501&pv127=127199&pv127=130342&pv127=130445&pv127=169384&pv127=169434&sf=1&FV=-8%7C183%2Cmu100mA%7C434%2Cmu10mA%7C434%2Cmu20mA%7C434%2Cmu25mA%7C434%2Cmu30mA%7C434%2Cmu3mA%7C434%2Cmu40mA%7C434%2Cmu45mA%7C434%2Cmu50mA%7C434%2Cmu60mA%7C434%2Cmu70mA%7C434%2Cmu75mA%7C434%2Cmu80mA%7C434%2Cmu85mA%7C434%2Cmu90mA%7C434&quantity=&ColumnSort=1000011&page=1&pageSize=25) - they work like optocouplers, or just a MOSFET.

It's hard to know what's happening, but I would try placing the filter on - this way you'll know if the 2000 values are deemed invalid. If they are valid, then it's likely a calibration issue.

WifWaf, 2020-11-14:
In regards to the square openings (vents), I think they're all for the same air  - it's just a route to pass through. I don't think there is a fan or anything like that, so it probably doesn't matter which way the air pass through.  Though I imagine as the sensor produces heat, it will create a heat gradient and hence airflow naturally.

I would though make sure they are not blocked, ust so air can pass freely through the whole device.

DerPicknicker, 2020-11-14:
Hello again 👋, 

thank you for your helpful answer. I will try to re-calibrate the sensor and I will enable the filter in the Code. 
Should I set the span to 2000? I remember that the values seemed to high. I will test it tomorrow Night and will response with some measurement.

Thank you for your Time.

DerPicknicker, 2020-11-16:
Hello, 

I made some measurements and I try to find the issues.. But I can’t figure them out. I attached the Files for you. 
For me the readings are quite good. But why are the Readings below 400ppm? 

[With Filter - no span - without new calibration - outside.txt](https://github.com/WifWaf/MH-Z19/files/5549050/With.Filter.-.no.span.-.without.new.calibration.-.outside.txt)
[With Filter - with span 2000 - with new calibration - outside.txt](https://github.com/WifWaf/MH-Z19/files/5549051/With.Filter.-.with.span.2000.-.with.new.calibration.-.outside.txt)
[With Filter - with span 2000 - without new calibration - outside.txt](https://github.com/WifWaf/MH-Z19/files/5549052/With.Filter.-.with.span.2000.-.without.new.calibration.-.outside.txt)

Best Greetings

WifWaf, 2020-11-16:
This means the CO2 levels are below calibration, it can be difficult to get it right without professional equipment.  

In practice, background CO2 can actually drop below 400pm also, as it's an average. If it's happening a lot though, it could be worth turning auto-calibration on, as it's dropping low enough. Alternatively, try to calibrate at nighttime and outside (when CO2 is at it's lowest).

Overall, it's only dropping a little bit, I wouldn't worry too much. You could add 10ppm to the final value, as atmospheric levels are closer to 410 than 400ppm.

DerPicknicker, 2020-11-18:
Hello.. sorry for my late response. 

In the meantime I made some test and can confirm that the first reading or the last reading is set to the background CO2 Level. I can’t understand this, because I calibrate the sensor and it looks to me if the sensor forgets the calibration... 

Is it needed to set the span to 2000? I made this in the setup function. 

I hope you can get my point and you could explain why the background co2 changes..

WifWaf, 2020-11-18:
The span only needs to be set when calibrating (2000 is recommended, but depends on the range), but it needs to sent in the [correct calibration order](https://github.com/WifWaf/MH-Z19/blob/master/examples/Calibration/Calibration.ino).

In non-calibration sketches, only autoCalibration(false) is needed to be sent in the setup.

Is this what you mean? 

[If possible, could you run this example, and copy the results here](https://github.com/WifWaf/MH-Z19/blob/master/examples/RetrieveDeviceInfo/RetrieveDeviceInfo.ino)?

DerPicknicker, 2020-11-18:
Ah okay... I found an issue in my code... I want to set the range in the setup() but I set the Span.. this could occur my problems.. 

I will remove this line and I will try your example. Is it possible that this set.span(2000)line could be the problem for background co2 levels?

WifWaf, 2020-11-20:
Just to clarify, is the issue the readings are appearing below 400ppm? 

It's not uncommon for CO2 to drop below 400ppm, but yes, calibration is always a factor. It might be worth turning autocalibration on, as the CO2 levels are likely dropping low enough. You can also use getCO2(false), but it just visually limits the reading from dropping below 400ppm.

Range refers to the range of CO2 you're measuring, span is the difference between the highest and lowest range. It's best to keep them at 2000 as the sensor is far less accurate above this and they work out to be the same in practice.

DerPicknicker, 2020-11-21:
Thank you for the explanation.. Sometimes the CO2 drops below 400 but not so often anymore. I can’t use the auto Calibration because the sensor won’t get to 400ppm every 24hours it’s used in a room and it is already calibrated.
I test your example code ..
This is the output:

Firmware Version: 04.43
Range: 2000
Background CO2: 410
Temperature Cal: 35
ABC Status: OFF

WifWaf, 2020-11-21:
**I've updated the library yesterday which is related to the autoCalibration(false)  - be sure to update your version as the bug can cause the sensor to crash, in particular on your firmware version**

Great to hear it's working better though, sounds like the calibration is in a good state. I plan to simplify the calibration in the future (in the testing branch).

DerPicknicker, 2020-11-22:
Now I updated the Libary and i wont only get 2000ppm as a value.. This cant be true... I cannot find the issue... The Temps are correct but the CO2 Values are way too high.. How can i fix that? I dont want to calibrate the Sensor every day. Any Ideas? 

The Backround PPM looks good : 
Firmware Version: 04.43
Range: 2000
Background CO2: 410
Temperature Cal: 35
ABC Status: OFF

Should I send the Code to you?

WifWaf, 2020-11-22:
Yes, send me the code, I'll have a look.

Also if you can let me know which device you are using to connect to the sensor.

DerPicknicker, 2020-11-22:
Thank you for your quick response... I attached an zip-File with my Code.. The Sensor is connected to an ESP32... And I use an ds3231 RTC to get an correct time for logging but also to get every 2 seconds an measurement of the sensor.. Sorry the code is quite long but its for debugging.. 
[mh-z19b-ds3231.zip](https://github.com/WifWaf/MH-Z19/files/5580289/mh-z19b-ds3231.zip)

WifWaf, 2020-11-22:
Looks ok at a glance.

Yes, that's normal - CO2 is relatively heavy, so sticks around for some time.

"Now I updated the Libary and i wont only get 2000ppm as a value.."

 It's possible that CO2 levels are that high if you are in front of the sensor, as CO2 builds up fast. Even 3000 - 4000 ppm is not uncommon with poor ventilation. 

I suspect these readings are correct, as they drop to the perfect range 400 - 500ppm by a window :).

WifWaf, 2020-11-22:
https://www.kane.co.uk/knowledge-centre/what-are-safe-levels-of-co-and-co2-in-rooms

DerPicknicker, 2020-11-22:
Okay thank you for the update. I will calibrate the sensor again and log some data... 
How often would you suggest to calibrate the sensor again? 

The autocalibration I not useable for me because the sensor should work in my office and it won't get every 24 hours fresh air to autocalibrate the sensor. 

Have a great evening.

WifWaf, 2020-11-22:
If the office is vacant at night, it should be fine to leave auto-calibration on.  For manual calibration, I tend to recalibrate the sensor once a year, though for most projects it's fine to calibrate once.

If you want to see higher values of CO2, you can set the range/span higher (maximum is 10,000), though I would personally try to keep both to 2,000, maybe 4,000 at most - it requires recalibration to change, however.

DerPicknicker, 2020-11-23:
After I did the calibration this morning the Readings are correct. And I would assume that the library don’t have any bugs. Thank you for your Help. My Co2-Measurement-Thing should be powered with an battery .. and the Sensor is only on when I am in the room.. That’s why its kind of tricky ;)

closed 2020-11-24 by DerPicknicker

WifWaf, 2020-11-24:
Glad it's working and no problem! Sensors are often a pain to validate - best of luck with the project (indoor CO2 levels are under-rated in terms of health importance :)).