WifWaf/MH-Z19 #23: Storage Usage of this Library ..

DerPicknicker, 2020-11-26:
Hello again 👍 , 

New Week, New Problem.. You know that I try to make an battery powered Device for my office.. So I would like to Use an Attiny.. But the Attiny 404 has 4kb of Space.. The Library with the BasicUsage won’t fit on that chip. 
The attiny has a real serial connection and don’t need „SoftwareSerial“.. Only for understanding: 8kb Attiny has only 26% free Space for Code. This must be more. 

How can I use this Libary without SoftwareSerial?

WifWaf, 2020-11-26:
Hey,  if it uses Arduino, yes.

You need to declare a Serial instance from the default Arduino libraries and pass it into the begin function once it's initiated. If you look at the examples, you'll see an ESP32 comment regarding its hardware serial, it will likely work like this.

Without seeing the Arduino package behind the Attinty 404 however, it's hard to be more specific. Do you have a link to a board?

DerPicknicker, 2020-11-26:
Hey, 

That Type is Not native from arduino.. it uses an Core like the esp32 Boards. 

Here is the link to the Core for supporting Attinys in the arduino Library:

https://github.com/SpenceKonde/megaTinyCore

WifWaf, 2020-11-26:
Ah,

Not entirely sure I'm right here as I can't look at this for long,[ but looks like the datasheet shows a single UART.](https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny202-04-402-04-06-Auto-DataSheet-DS40002159A.pdf)

Assuming this is the correct header for the board, [it also looks like the hardware serial is only declared once](https://github.com/SpenceKonde/megaTinyCore/blob/aba2768259aa1cb19bf43ffb0392e0b955249b6f/megaavr/variants/txy4/pins_arduino.h#L61).

[However, the UART library can expect up 4.](https://github.com/SpenceKonde/megaTinyCore/blob/aba2768259aa1cb19bf43ffb0392e0b955249b6f/megaavr/cores/megatinycore/UART.h#L224).

This means you need to use SoftwareSerial if you plan to use the Serial for a terminal or uploading code - double check with the designer of the module.

DerPicknicker, 2020-11-26:
> Ah,
> 
> 
> 
> Not entirely sure I'm right here as I can't look at this for long,[ but looks like the datasheet shows a single UART.](https://ww1.microchip.com/downloads/en/DeviceDoc/ATtiny202-04-402-04-06-Auto-DataSheet-DS40002159A.pdf)  (page 10).
> 
> 
> 
> Assuming this is the correct header for the board, [it also looks like the hardware serial is only declared once](https://github.com/SpenceKonde/megaTinyCore/blob/aba2768259aa1cb19bf43ffb0392e0b955249b6f/megaavr/variants/txy4/pins_arduino.h#L61).
> 
> 
> 
> [However, the UART library can expect up 4.](https://github.com/SpenceKonde/megaTinyCore/blob/aba2768259aa1cb19bf43ffb0392e0b955249b6f/megaavr/cores/megatinycore/UART.h#L224).
> 
> 
> 
> This means you need to use SoftwareSerial if you plan to use the Serial for a terminal or uploading code - double check with the designer of the module.

I checked it but I don't need serial for uploading code. One pin is used for uploading code (UPDI).. So I think that I can use the (hardware) serial Connection.

WifWaf, 2020-11-26:
I can't read it all through, unfortunately - I would guess you could just write myMHZ19.begin(Serial), as the base class (Stream) is inherited from the UartClass they've built on top of the HardwareSerial class.

I would ask them however if you're not sure as it's not really an issue for this library.

By the way, if you look at the testing branch, there is a version of this library that uses significantly less memory - though as indicated by the way, it hasn't been thoroughly tested. It has updated examples, however.

DerPicknicker, 2020-11-26:
I edit the Setup like this: 

void setup()
{
    Serial.begin(9600);                                     // Device to serial monitor feedback

   // mySerial.begin(BAUDRATE);                               // (Uno example) device to MH-Z19 serial start   
    //mySerial.begin(BAUDRATE, SERIAL_8N1, RX_PIN, TX_PIN); // (ESP32 Example) device to MH-Z19 serial start   
    myMHZ19.begin(Serial);                                // *Serial(Stream) refence must be passed to library begin(). 

    myMHZ19.autoCalibration();                              // Turn auto calibration ON (OFF autoCalibration(false))
}

Now the Compilation works.. Software Serial needs so much space... After removing SoftwareSerial it won’t fit. But after using your Testing-Branch the Code fit to 4kb.. 3415 Bytes are used (83%).. I think it could be improved but for me important is that it fits. When my Attiny arrives I will test it.

WifWaf, 2020-11-26:
Very nice! Yes, it does take a lot of space... let me know if you have any issues with the testing branch version.

DerPicknicker, 2020-11-26:
Yep i will do. I only need to find out how to specify the Serial-Pins... I will contact the maintainer of megaTinyCore..

WifWaf, 2021-01-13:
Closed for now, feel free to open if any more issues.

closed 2021-01-13 by WifWaf