WifWaf/MH-Z19 #26: Get / Set - Cycle

SFeli, 2020-12-23:
Hello WifWaf,
on https://revspace.nl/ there is a indicator how to change the cycle of MH-Z19b.
Default is 5 seconds which will also be responded by : 
int MHZ19::getCycle()
{
    /* check get Cycle length (5 Sek default) */
    provisioning(GETCYCLE);
    if (this->errorCode == RESULT_OK)
        /* convert MH-Z19 memory value and return */
        return (int)makeInt(this->storage.responses.STAT[2], this->storage.responses.STAT[3]);
    else
        return 1;
}

But I'm not fine with setting other values.
Did you have checked this register (126) already?

If you like to investigate - I have setup some lines in a fork-version.
Nice to get your ideas and have a nice time . You see I'm a longeterm follower.

Stefan

WifWaf, 2020-12-26:
Hey Stefan,

Thanks for the info :).

I had noticed my sensor had a 5 written at the register - I had no idea what it meant though! Upon writing to the register, it changed to 1 - but that's all I got from it unfortunately.

I have an Arduino serial program for experimenting with the sensor, though it was one of my first pieces of code - consequentially, it's a bit poorly written and with bugs.

If you can e-mail at:  jdwifwaf@gmail.com, I'll send it over to you as I still find it very useful.

Jon

WifWaf, 2020-12-28:
Just to add, [if you check my scrappy notes and click on tab 149 ](https://docs.google.com/spreadsheets/d/1hSbtUwD5b78hpo37Z1yIxQ3oiaQXUNfCuivmhBwS0-E/edit?usp=sharing), there seems to be a command with an array of read-only values behind it.

3 of which might be date code? Also, a temperature value with (what looks like) an 8 bit decimal place, is found. The rest I'm not sure about...

See what you think (I'm hoping to have more time soon to experiment with the sensor)

WifWaf, 2021-01-25:
Feel free to re-open / e-mail.

closed 2021-01-25 by WifWaf