WifWaf/MH-Z19 #27: FYI: be warned about fake banggood MH-Z19b sensors

hixfield, 2020-12-28:
I recently ordered 3x MH-Z19 sensors from banggood. Because I wanted to construct 3 supplementary IOT CO2 sensors (refer to my project https://github.com/hixfield/HixCO2Tem...). However, the readings that they produce are highly unstable and cannot be trusted. I created small YouTube video to show my readings (on request of banggood itself):

https://youtu.be/5_QQe75-SZI

Also found this post about it: https://revspace.nl/MH-Z19B#Fake_MH-Z19B_.28black_PCB.29

WifWaf, 2020-12-28:
Cheers for the information - it's the first I've heard of this.

If it's okay with you, I'll add a note to the Readme with a link to your video.

hixfield, 2020-12-28:
Yes sure, excellent idea!

WifWaf, 2020-12-28:
Done. I'll leave it up for as long as they are in circulation, then move it lower down the readme.

Cheers for the info!

closed 2020-12-28 by WifWaf