WifWaf/MH-Z19 #32: Disbale ABC each day?

bytesontherocks, 2021-01-27:
Hello, 

This is just a quick question:

In the project Readme says "ABC must be disabled each day, however this is handled by the library."
Is this correct? So the MHZ19B re-enables the auto cal every 24h? Could you confirm it, please?

Many thanks.

WifWaf, 2021-01-27:
Hey, the library re-sends the ABC OFF command every 12 hours, which is checked whenever a function is called.

Essentially, to keep auto-calibration OFF, the ABC OFF command must be sent before 24 hours elapses, otherwise it re-enables.

bytesontherocks, 2021-02-01:
Hi, thanks for your reply.

Using my own code as a test I disabled the auto cal in an mhz19b sensor. I'm reading the status of the auto cal logic every 5s, and I have been doing it for more than 30h. I can confirm that the auto cal hasn't been re-enabled so I don't see what you have described. 

Could it be some change in the sensor firmware? I'm using a 2020 sensor, but I don't have the firmware version.

WifWaf, 2021-02-01:
Ah, I'm with you now - yes, it was from an older datasheet (though dubious translation) - it's been omitted from the latest datasheets.

I remember it irritating people with earlier sensors (this could possibly be MHZ19 (not B)) as it would cause drops in ppm every 24 hrs. The ABC check byte wasn't known at the time, so I've never actually confirmed it; It would be interesting to confirm with an early sensor.

If it's not being turned off, then there's no harm in removing the check: https://github.com/WifWaf/MH-Z19/blob/e8fa93ea54739e2ce3502b269ffe5a38a4778364/src/MHZ19.cpp#L474

Hopefully, I'm on the same page now.

closed 2021-02-22 by WifWaf