WifWaf/MH-Z19 #33: error compiling

wasteoinc, 2021-02-01:
Hi, thanks for the library, makes life so much easier setting up the device

I tried the basic sketch with both library files inside the same directory and tried to compile it on an arduino pro mini (ATMEGA328 so uno compatible) and im getting the following error

```
MHZ19.cpp: In member function ‘byte MHZ19::read(byte*, MHZ19::Command_Type)’:
MHZ19.cpp:594:48: error: invalid conversion from ‘byte* {aka unsigned char*}’ to ‘char*’ [-fpermissive]
     mySerial->readBytes(inBytes, MHZ19_DATA_LEN);
                                                ^
In file included from /usr/share/arduino/hardware/arduino/cores/arduino/HardwareSerial.h:28:0,
                 from /usr/share/arduino/hardware/arduino/cores/arduino/Arduino.h:193,
                 from MHZ19.h:6,
                 from MHZ19.cpp:3:
/usr/share/arduino/hardware/arduino/cores/arduino/Stream.h:76:10: note:   initializing argument 1 of ‘size_t Stream::readBytes(char*, size_t)’
   size_t readBytes( char *buffer, size_t length); // read chars from stream into buffer
          ^
```

have i fucked up something obvious that i can't see?

WifWaf, 2021-02-01:
I can compile for the Pro Mini with both hardware and software serial - odd.

It should be passed to this function instead: https://github.com/arduino/ArduinoCore-avr/blob/9f8d27f09f3bbd1da1374b5549a82bda55d45d44/cores/arduino/Stream.h#L97

Is your Arduino core up to date? Are you using your own core? Also, could you paste your setup code?

wasteoinc, 2021-02-02:
I hate different compiler versions :(
As long as you said it it clicked and it's not even the first time it happens to me
my arduino ide was 1.0.5 that came together with my linuxmint
1.8.13 compiles correctly 

thanks for all the fishes

WifWaf, 2021-02-02:
Great stuff - happens to the best of us too often!

closed 2021-02-02 by WifWaf