WifWaf/MH-Z19 #34: get background CO2

wasteoinc, 2021-02-02:
Now that I finally experiment I have another question.

I did a manual calibration as you suggested in your examples. I left the device outside for 20-30 minutes and then I pulled the DH pin to ground for 7 seconds. The readings calibrated to 400 and then i let it run for another 10 minutes to see if it's stays there.

After that i placed the device inside a closed room to show the higher readings. It did willfully. But then i checked the getBackgroundCO2 commend and I noticed the following.

The getBackgroundCO2 was always printing the same as the CO2 command

Background CO2: 3067
ABC Status: OFF
CO2 (ppm): 3067
Temperature (C): 20

Background CO2: 3100
ABC Status: OFF
CO2 (ppm): 3100
Temperature (C): 20

If I got the documentation correctly the getBackgroundCO2 should have been a constant 400 after the calibration. Have you any idea what am i missing?

WifWaf, 2021-02-02:
This is not your error, nor the senors.

Newer firmware versions (which are hard to keep track of) have replaced the registry entry with the last CO2 reading taken (or so it appears).

I have an idea of how to request the data from all versions, but until then, I plan to remove the function from the working examples. 

Hopefully, this helps?

wasteoinc, 2021-02-03:
of course it does, thanks

closed 2021-02-04 by WifWaf