WifWaf/MH-Z19 #36: Wrong Temperature - wrong Co2?

5yF0Rc3, 2021-03-21:
Hi, I used your example script on my new (and first) MH-Z19B sensor. I have no experience with this type of Sensor, however i do not trust this temerature measurements:
`CO2 (ppm): 1962
Temperature (C): 4
`
The Co2 could be possible due to an indoor room, but the temp is definitely off. And I guessing this makes the Co2 wrong too? Could this be script side oder is my sensor just broken?

WifWaf, 2021-03-21:
What temperature were you expecting?

Also which MCU are you using, and does your sensor have a black or green PCB?

Cheers,

5yF0Rc3, 2021-03-21:
Thanks for your reply. Its about 22 - 24 Deg Celcius in my room, the temp is stable at 3-4 Deg. But the sensor does "work", holding it out of my window shows about -7 Deg (should be about 3), probably going to sink some more if its longer outside. 
Im using a real green PCB one.

WifWaf, 2021-03-21:
Could you run the "RetrieveDeviceInfo" sketch, and paste the version number here? 

It's possible it's due to a firmware update (as your sensor is new) which lack releases notes.

5yF0Rc3, 2021-03-21:
`Firmware Version: 05.00`
`Range: 2000`
`Background CO2: 1565`
`Temperature Cal: 35`
`ABC Status: OFF`

WifWaf, 2021-03-22:
Thanks, looks like a new version.  

Does getTemperature(true) return the same/similar value? 

Also, it shows auto-calibration is OFF, is this correct?

5yF0Rc3, 2021-03-22:
Yes, getTemperature(true) returns the same result. And Yes I toggled the ABS off. While playing with the parameters i also noticed that the .getCO2(true,true) still gives me the limited ppm of 2000 as a maximum...

WifWaf, 2021-03-23:
Thanks for the info. 

As the temperature value is not linear, I'm not sure if there is a way to get around it at the moment. However, on the plus side, your sensor appears to be working correctly.

Just co clarify are using the MH-Z19B, not the MH-Z19C?

5yF0Rc3, 2021-03-23:
Yes im using the B version. I'll try the sensor in diffrent temperature environments to test the range of the sensor (I think it could still be linear). However, while testing I stumbled about the following problem:

**.getCO2(true,true)**
`Sent << DEC: 255 1 133 0 0 0 0 0 122  `
`Recieved >> DEC: 255 133 5 43 3 43 5 114 166  `
`CO2 (ppm): 811`
`Temperature (C): 3`

for the Co2; 5*256 + 43 = 1323ppm 

**.getCO2(false,true)**
`Sent << DEC: 255 1 134 0 0 0 0 0 121  `
`Recieved >> DEC: 255 134 3 90 42 0 0 0 243  `
`CO2 (ppm): 858`
`Temperature (C): 3`

for the Co2; 3*256 + 90 = 858ppm 

why does the Command 133 recieve a different ppm, but in the script returns the one from 134 ? I dont understand your lib good enought to fully understand the read function.

WifWaf, 2021-03-23:
Yep, that's correct - slightly different protocol format for returning CO2 values.

Yes, 133 should be unclamped. It's a bit of guessing as to what's been changed at the moment. Out of curiosity, when you call getBackgroundCO2(), are you receiving the same value?

If you can confirm the temp value is still linear, try setting the [temp to adjust to 20 in the header file](https://github.com/WifWaf/MH-Z19/blob/e8fa93ea54739e2ce3502b269ffe5a38a4778364/src/MHZ19.h#L14) (the default is supposed to be 40).

5yF0Rc3, 2021-03-24:
Yes, the getBackgroundCO2() is the same.

brozikcz, 2021-03-26:
I have a similar issue with my MH-Z19C, firmware version 05.12.
The temperature random jumps about +-5 degrees celsius and the background command return similar values as the get CO2.

WifWaf, 2021-03-26:
Cheers for the info.

 I'm not sure if it's possible to uncap the CO2 and fix the temperature values on the new firmware just yet.

I recently picked up the MH-Z19C, so will hopefully know a bit more soon...

brozikcz, 2021-03-26:
Could I help somehow with debugging?

5yF0Rc3, 2021-03-27:
> I'm not sure if it's possible to uncap the CO2 and fix the temperature values on the new firmware just yet.

hmm. Can't we just set the Range higher? If the range is 5000, up to 2000 should be as accurate as if the range is 2000... or not?

WifWaf, 2021-03-27:
The range is used in the CO2 ppm calculations and changing it will lose accuracey.

To get around this, you need to provide a reference ("zero") for the span so it knows what 5000 ppm looks like; this needs calibration equipment.

WifWaf, 2021-03-27:
> Could I help somehow with debugging?

Thanks for the offer. 

Figuring out what's changed and how to use the new values is the goal. I do have a "serial tool" for helping with this, however, it's a bit awkward and poorly coded (I can still send it jdwifwaf@gmail.com). 

It is possible to download the firmware from the device too with a j-link.

stefan123t, 2022-02-16:
@WifWaf I am interested in downloading the firmware using a ST-Link v2.
I have logged #47 to document the necessary steps for that.

WifWaf, 2022-07-28:
This should have been fixed in https://github.com/WifWaf/MH-Z19/commit/04a6f85f980cae100f6c9472a1bbe484cc10dd3c.

Please see this issue for further comments https://github.com/WifWaf/MH-Z19/issues/43

closed 2022-07-28 by WifWaf