WifWaf/MH-Z19 #37: Readings way off

marcelstoer, 2021-04-20:
This is likely related to #36 but I did not want to hijack that thread.

I have a recent MH-Z19C (plug version). The C02 readings have been ok-ish from the beginning. I compare them to the readings I get out of a Sensirion SCD41 that sits next to it. The temperature was off by roughly +10°C.

However, after a few weeks the CO2 readings started declining gradually until they ended up far below 400ppm (120-300). The reported temperature was constantly around +-32°C.

To see whether this library may be at fault I tried two others for this sensor and got similar results. I concluded my sensor must have failed the test of time. However, after I scanned #36 I'm not so sure anymore.

Tonight I ran the sensor outside for half an hour. Here's what it reports now.

**Device info**
```
21:57:22.753 -> Firmware Version: 05.12
21:57:22.753 -> Range: 2000
21:57:22.787 -> Background CO2: 500
21:57:22.787 -> Temperature Cal: 50
21:57:22.823 -> ABC Status: ON
```
**Basic usage plus background CO2**
```
22:18:26.583 -> CO2 (ppm): 493
22:18:26.583 -> background CO2: 493
22:18:26.654 -> Temperature (C): 33
22:18:28.653 -> CO2 (ppm): 497
22:18:28.653 -> background CO2: 497
22:18:28.725 -> Temperature (C): 33
22:18:30.716 -> CO2 (ppm): 469
22:18:30.716 -> background CO2: 469
22:18:30.782 -> Temperature (C): 33
22:18:32.815 -> CO2 (ppm): 465
22:18:32.815 -> background CO2: 465
22:18:32.850 -> Temperature (C): 33
22:18:34.858 -> CO2 (ppm): 465
22:18:34.858 -> background CO2: 465
22:18:34.895 -> Temperature (C): 33
22:18:36.939 -> CO2 (ppm): 468
22:18:36.939 -> background CO2: 468
22:18:36.976 -> Temperature (C): 33
22:18:38.985 -> CO2 (ppm): 466
22:18:38.985 -> background CO2: 466
22:18:39.056 -> Temperature (C): 33
22:18:41.054 -> CO2 (ppm): 491
22:18:41.054 -> background CO2: 491
22:18:41.126 -> Temperature (C): 33
22:18:43.143 -> CO2 (ppm): 492
22:18:43.143 -> background CO2: 492
22:18:43.181 -> Temperature (C): 33
22:18:45.198 -> CO2 (ppm): 485
22:18:45.198 -> background CO2: 485
22:18:45.232 -> Temperature (C): 33
22:18:47.263 -> CO2 (ppm): 436
22:18:47.263 -> background CO2: 436
22:18:47.299 -> Temperature (C): 33
22:18:49.342 -> CO2 (ppm): 440
22:18:49.342 -> background CO2: 440
22:18:49.380 -> Temperature (C): 33
22:18:51.414 -> CO2 (ppm): 494
22:18:51.414 -> background CO2: 494
22:18:51.450 -> Temperature (C): 33
22:18:53.455 -> CO2 (ppm): 506
22:18:53.455 -> background CO2: 506
22:18:53.529 -> Temperature (C): 33
```

-> CO2 and background always the same
-> CO2 is off by ~500ppm compared to the SCD41 that I consider accurate
-> CO2 readings are highly volatile, jumping 10% in just a couple of seconds
-> temperature is off by 10°C

WifWaf, 2021-04-20:
Yes, it's likely the same issue.

The gradual decline in ppm would be due to the sensors autocalibration.  Why it is going below 400ppm I'm not sure? While it could be correct (it does occur sometimes), I agree it's unlikely if your other sensor is reporting differently. If I was to guess, it could be a range issue, you could try setting the range to 2000 and see if it helps (changing span needs calibration equipment). If this fails, you could try zeroing the sensor manually and turning autocalibration off.  

I hope to have some time to look at the new firmware changes next week (or the week after), I have the C version to test.

WifWaf, 2021-04-20:
Just to add, the background CO2 value mirroring get CO2 is also due to the new firmware.  It's not usually volatile however unless you are near to it (breathing etc) or have the black pcb version.

marcelstoer, 2021-05-01:
I've been using a 2nd MH-Z19C for the past 10d (continuously). Today I grew suspicious if its readings; have them visualized on a screen on my desk.  Sure enough, when compared to the Sensirion SCD41 it's again off by a few hundred ppms. That wasn't the case when I started using it.

Even though you provide great sketches incl. explanations I haven't tried fiddling with the settings of the sensor yet for a couple of reasons:
- I am afraid I might brick it completely
- I do not fully understand what I am doing
- As this driver doesn't _explicitly_ support the C-version we can't be sure running any "special" commands (tweaking settings) will work as intended.

Is it justified I am worried I could irreversibly break something running your sketches? Given the described behavior what procedure would you recommend to either fix it or to provide maybe more data points?

WifWaf, 2021-05-11:
This looks correct - if you're using auto-calibration, you should be able to leave it on here as the calibration (zeroing) function isn't being called.

During calibration testing, I have had to reset the span a few times. It's not clear why or whether it was due to the sensor I was using (I corrupted the firmware), however, it 'shouldn't' be needed.

WifWaf, 2021-07-23:
Closing for now as no activity, feel free to comment again to open the issue.

closed 2021-07-23 by WifWaf