WifWaf/MH-Z19 #38: does not work on arduino mega

darksh3ll, 2021-06-20:
works perfectly on ARDUINO UNO but on ARDUINO MEGA the probe is not recognized

WifWaf, 2021-06-20:
Just given it a test and it's working OK for me on the mega.

Can you paste your setup code? Cheers,

darksh3ll, 2021-06-20:
thank you for your reply here is the code 

`
#include <Arduino.h>
#include "MHZ19.h"
#include <SoftwareSerial.h>
#define RX_PIN 10
#define TX_PIN 11
#define BAUDRATE 9600

MHZ19 myMHZ19;
SoftwareSerial mySerial(RX_PIN, TX_PIN);

unsigned long getDataTimer = 0;

void setup() {
  Serial.begin(9600);
   mySerial.begin(BAUDRATE);
   myMHZ19.begin(mySerial);
  myMHZ19.autoCalibration();
}

void loop() {

      static int CO2=0;
    if (millis() - getDataTimer >= 2000)
    {
        CO2 = myMHZ19.getCO2();
        Serial.print("CO2 (ppm): ");Serial.println(CO2);
        getDataTimer = millis();
    }

  delay(500);

}
`

WifWaf, 2021-06-20:
Ah, this library hasn't been tested with the SoftwareSerial library on the Mega as it has 3 hardware serial peripheral ports.
 
 If available, use Serial1 (which is attached to TX1 and RX1, pins 18 and 19 respectively), and setup as below:

<pre><code>#include <Arduino.h>
#include "MHZ19.h"                                        

#define BAUDRATE 9600

MHZ19 myMHZ19;

unsigned long getDataTimer = 0;

void setup()
{
    Serial.begin(9600); 
    Serial1.begin(BAUDRATE);  
    myMHZ19.begin(Serial1);  
    myMHZ19.autoCalibration(); 
}</code></pre>

WifWaf, 2021-07-23:
Closing the issue for now as it is related to the software serial library.

closed 2021-07-23 by WifWaf