WifWaf/MH-Z19 #40: Simplification proposal

etodorovich, 2021-06-29:
In MHZ19.cpp, sections such as: 
```
        #if defined (ESP32) && (MHZ19_ERRORS)
        ESP_LOGE(TAG_MHZ19, "message...");
        #elif MHZ19_ERRORS
        Serial.println("!ERROR: message...");
        #endif 
```

Could be simplified:
```
        #if MHZ19_ERRORS
        Serial.println("!ERROR: message...");
        #endif 
```
Thank you!

WifWaf, 2021-06-29:
ESP logging uses ANSI escape codes which can be useful for serial monitor/terminal colours, though I have wondered whether it's clashing with some Serial prints (ESP log always includes CR LF, Serial.print does not)

I don't think this qualifies as an issue as such, as it does not affect performance, but feel free to submit a pull request for future versions. 

If you're looking through the code, you might also want to check out the test branch [PR_.2.0.0](https://github.com/WifWaf/MH-Z19/tree/PR_2.0.0/src) which is more recent, including an overhaul to reduce memory usage.

Cheers!

etodorovich, 2021-06-29:
Thank you very much. If it is possible to collaborate, count on me. We are using Version: 1.5.3. I will test the new versions of the library as soon as posible.

WifWaf, 2022-07-28:
Closing for now as I seem to have dropped the ball on this one?  Feel free to re-open.

closed 2022-07-28 by WifWaf