WifWaf/MH-Z19 #41: Temperature values among different sensors

homuhumushomu, 2021-07-23:
I built 4 devices, all with the same components including a CO2 sensor, all running the same arduino software. I put them side by side and they report wildy different temperature values (14, 24, 38, 45). Have you seen this problem? What should be this value?

WifWaf, 2021-07-23:
It's a compatibility issue with the newer sensor firmware changes - hope to have it fixed soon.

homuhumushomu, 2021-07-23:
Is there any information I can provide to help you?

homuhumushomu, 2021-07-25:
Does this affects the CO2 measurement? Because I am seeing a large variance among them a few days after calibration (~370, 410, 430, 445) but a rather small standard deviation within each (~ 5 ppm, 1000 measurements over from 0 to 3AM)

WifWaf, 2022-07-28:
Only a year late, but this should have been fixed in 04a6f85f980cae100f6c9472a1bbe484cc10dd3c.

closed 2022-07-28 by WifWaf