WifWaf/MH-Z19 #42: Library incompatible with I2C LCD?

luizedu91, 2021-10-09:
I've been using a different MHZ19 library but am having serious bugs with it so I gave yours a try, however, when I upload it the LCD doesn't work. Uploading the old sketch brings it back online. I can't understand how this would affect the screen, but I've changed nothing in my code but what's needed to make each library work.

I'm running a Leonardo, so using hardware serial, if that's relevant. 

Any help?

WifWaf, 2021-10-09:
It hasn't been tested on the Leonardo, however, it should work.

Any idea at which point in the sketch it's crashing? Are you able to see any prints on the serial monitor?

WifWaf, 2021-10-09:
Okay, unfortunately without any more details I can't determine the cause.

If you use the library in future, feel free to open another issue.

closed 2021-10-09 by WifWaf