WifWaf/MH-Z19 #44: ESP32C3 support.

roberbike, 2022-01-10:
Compiling for an esp32c3 board, I have some issues with platfomio, like this:

C:/Users/Usuario/.platformio/packages/framework-arduinoespressif32@src-01b2ee0664276ec022da2783d94579c3/tools/sdk/esp32c3/include/log/include/esp_log.h:351:38: note: in expansion of macro 'ESP_LOG_LEVEL_LOCAL'
 #define ESP_LOGE( tag, format, ... ) ESP_LOG_LEVEL_LOCAL(ESP_LOG_ERROR,   tag, format, ##__VA_ARGS__)
                                      ^~~~~~~~~~~~~~~~~~~
.pio/libdeps/ESP32C3/MH-Z19/src/MHZ19.cpp:51:9: note: in expansion of macro 'ESP_LOGE'
         ESP_LOGE(TAG_MHZ19, "Invalid Range value (0 - 65000)");
         ^~~~~~~~
.pio/libdeps/ESP32C3/MH-Z19/src/MHZ19.cpp:51:18: error: 'TAG_MHZ19' was not declared in this scope
         ESP_LOGE(TAG_MHZ19, "Invalid Range value (0 - 65000)");

WifWaf, 2022-01-10:
The TAG definition seems to have been lost in a previous update. The ESP32 doesn't seem to mind for some reason, so it went under the radar.

I've committed the change to the library: https://github.com/WifWaf/MH-Z19/blob/cdac62c0bfca06572521f2da22ffba99156e0adb/src/MHZ19.h#L10

I can't test it out on the C3 right now, but it should work as long as the compiler recognises it as an "ESP32".

Let me know.

WifWaf, 2022-02-16:
Hopefully, this is resolved now, if not, let me know.

closed 2022-02-16 by WifWaf

hpsaturn, 2022-06-06:
Thanks @WifWaf but maybe we need a pull request to the third party library owner. Maybe can you do it?