WifWaf/MH-Z19 #46: Question. Correct interpretation of sensor readings

LazarenkoA, 2022-02-07:
I am using version 1.5.3

Hi.
I calibrated on a windowsill with an open window, after calibration I upload my sketch and invoke getBackgroundCO2(), it was 500, then I corrected something in the code, upload it again and the value of getBackgroundCO2() became equal to getCO2(), is this correct behavior? I had the idea to calibrate the sensor, and then measure the delta from the background level of CO2, i.e. getCO2() - getBackgroundCO2(). Also, what seemed strange to me, the sensor readings in a ventilated room range from 1500 to 1600 ppm, is this an overestimated value?

WifWaf, 2022-02-08:
Newer firmware versions have created a few issues for the library. It previously held the assumed average background CO2 concentration (400 ppm). On newer versions, it's something similar to the current CO2 value. I'll likely remove the function in a future update as it's become more misleading than not.

In terms of the indoor CO2 value, I would agree it's a bit high. Typically, autocalibration works okay given a few weeks... manual calibration highly depends on the environment it was calibrated within, fresh air at night can work well.

LazarenkoA, 2022-02-08:
thanks for the reply. During auto-calibration, if the device is disconnected from the power supply, the results of auto-calibration are stored in it?

WifWaf, 2022-02-08:
Yes, it stores the values when the power is off and is usually more accurate over time.

If I understand correctly the lower threshold would be the lowest recorded CO2 value that the sensor uses for autocalibration. Technically, you should be able to get the value from the sensor, however, I'm not sure which registry it's held in or if it's in ppm.

If you turn auto autocalibration off when it's in a good state, it should retain those settings. Alternatively, you could generate an equation for the relationship between the raw values and CO2 values at this point, although it's a bit more involved and it's unclear if the 'raw' values are adjusted for temperature.

WifWaf, 2022-07-28:
Closing for now due to inactivity, feel free to reopen if you need any more info.

closed 2022-07-28 by WifWaf