WifWaf/MH-Z19 #47: Backup / Restore Firmware using ST-Link v2 and OpenOCD

stefan123t, 2022-02-16:
@WifWaf</a> I am interested in downloading the firmware using a ST-Link v2.<br>
I found the SWD ports on the schematics at <a href="https://revspace.nl/MH-Z19B" rel="nofollow">https://revspace.nl/MH-Z19B</a></p>
<h2 dir="auto"><span id="user-content-firmware">4. Firmware</span></h2>
<p dir="auto">MCU firmware can be easily dumped through SWD pads. Pinout:
</p>

Pin | Signal
-- | --
1 (square) | 3V3
2 | GND
3 | SWDIO
4 | SWCLK
5 | RESET (active low)

<p dir="auto">Additionally, bootloader asks for a firmware update during the first 20s of startup.</p>

[image]

WifWaf, 2022-02-16:
Cheers for the info. I've downloaded the firmware previously also, however, I couldn't get it to upload successfully. Have you had success here?

I know little about reverse engineering firmware, so wasn't able to get any insight.

stefan123t, 2022-02-17:
Well I am new to the topic either currently stuck with the download / connect part. But once you have the binary file you could use some disassembler like Ghidra/IDA Pro to make it assembler or C code again.

stefan123t, 2022-02-18:
My MH-Z19C is built on 2021 01 04 and differs a bit from the MH-Z19B pictures posted on the revspace.nl page.
```
$ openocd -f interface/stlink.cfg -f target/stm32f1x.cfg
Open On-Chip Debugger 0.11.0+dev-00571-g254883597 (2022-02-17-20:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : DEPRECATED target event trace-config; use TPIU events {pre,post}-{enable,disable}
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J38S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.157485
Warn : UNEXPECTED idcode: 0x0bf11477
Error: expected 1 of 1: 0x1ba01477
```

Actually mine does not contain an STM32F104C8T6 anymore but it most likely contains a GigaDevice GD32e23x CPU (`idcode: 0x0bf11477`). At least I could get openocd to run without any error using this chip target: 
`openocd -f interface/stlink.cfg -f target/gd32e23x.cfg`

```
$ openocd -f interface/stlink.cfg -f target/gd32e23x.cfg
Open On-Chip Debugger 0.11.0+dev-00571-g254883597 (2022-02-17-20:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J38S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.157485
Info : gd32e23x.cpu: Cortex-M23 r1p0 processor detected
Info : gd32e23x.cpu: target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for gd32e23x.cpu on 3333
Info : Listening on port 3333 for gdb connections

```