WifWaf/MH-Z19 #48: MH-Z19B measures well the first 15 minutes, then 0 ppm

Osfuco, 2022-02-25:
Good afternoon, I have an MH-Z19B connected to an ESP12, when I connect it to the power supply it works fine for the first 15 minutes, then it stops measuring 0 ppm. Does anyone know how to fix it?

WifWaf, 2022-02-28:
Have you tried manually calibrating the sensor? It might be due to the sensor lacking any reference values.

Outside of this, the only time it's happened to me is when I corrupted the firmware, but it took quite some effort.

Osfuco, 2022-02-28:
The sensor is calibrated. In fact, for the first few minutes it works perfectly, but then it dies (the light keeps flashing but it sends 0ppm).

There are 3 sensors. Maybe they are defective?

Osfuco, 2022-02-28:
I have a 4th sensor model MH-Z19C with the same code and works perfectly all the time (15 days)

WifWaf, 2022-03-01:
Is the issue with all 3 sensors?

If the library filter is on, it might be worth turning it off and seeing if it zeros or just erroneous data.

As a last resort, you could try setting the range to 2000 (if that's the CO2 limit you using), although it might make things worse.  I can't remember the reason, however, I've had to do with this with a sensor before. Unfortunately, it's been a while since I used the sensor.

WifWaf, 2022-07-28:
Closing for now due inactivity, feel free to reopen.

closed 2022-07-28 by WifWaf