WifWaf/MH-Z19 #49: ERROR 3 after powercycle

Knochi, 2022-02-27:
Hi,

so i have an ATTiny1614 (16kB Flash, 2kB RAM) with megaTinyCore, a 4-digit serial display and the MHZ-19C. I'm using hardware serial to save memory.
When i flash the code everything works as expected but when i power cycle the board i get error 3 "Recieved data does not match the usual syntax expected".
Flash memory is at 31% and global variables consume 10%.

If it is a memory problem, why does it work right after flashing?

```C
#include <Arduino.h>
#include "MHZ19.h"
#include <TM1637Display.h>

//Pins for 7-Segment
#define CLK 7
#define DIO 6

MHZ19 myMHZ19;                    // Constructor for CO2 Sensor
TM1637Display display(CLK, DIO);  // Constructor for 7-Segment

unsigned long getDataTimer = 0;
uint16_t errorCode=0;

void setup()
{
    display.setBrightness(0x05);
    Serial.begin(9600);                                     // Device to MH-Z19 should not be changed
    
    //mySerial.begin(BAUDRATE);                               // (Uno example) device to MH-Z19 serial start   
    myMHZ19.begin(Serial);                                // *Serial(Stream) refence must be passed to library begin(). 
    myMHZ19.printCommunication();    
     
    myMHZ19.autoCalibration(true);                              // Turn auto calibration ON (OFF autoCalibration(false))
    display.showNumberDec(millis(),false);
    delay(2000);
    display.showNumberDec(0,true);
}

void loop()
{   
    if (millis() - getDataTimer >= 2000)
    {
        int CO2; 

        /* note: getCO2() default is command "CO2 Unlimited". This returns the correct CO2 reading even 
        if below background CO2 levels or above range (useful to validate sensor). You can use the 
        usual documented command with getCO2(false) */

        CO2 = myMHZ19.getCO2();                             // Request CO2 (as ppm)
        if(myMHZ19.errorCode ==  RESULT_OK)
          display.showNumberDec(CO2,false);
        else{
          display.showNumberDec(myMHZ19.errorCode,true);
        }
        getDataTimer = millis();
    }
}
```

WifWaf, 2022-02-28:
It sounds as if the UART data is going out of sync, however, the library should handle this. Have you anyway to check the UART signal integrity?

If not, have you tried the sensor on another microcontroller? I'm not familiar with ATTiny series.

Knochi, 2022-05-02:
So i think i found the problem.
First of all, because _printCommunication()_ is "true,true" by default it introduces a lot of additional, unseen, traffic to the serial bus when using the hardware serial. So I switched that off. But i don't think this introduces any problems.

The second finding is more important. Setting a single shot trigger on my logic analyzer and tracing the first bus communication showed some traffic **from** the Sensor and with 115200 baud. Then i hooked up a serial2USB with the RX port connected to the sensors TX and voila.
[image]

The Sensor starts up with an autoload firmware routine, that is lasting about 30s. So I would propose a routine in the init function of the sensor library that catches this and sends an 'a' to abort this.

WifWaf, 2022-05-02:
Cheers for the info! If I am understanding correctly, the serial output is from the sensor on start-up? Have you tried sending 'a' to abort it? Also, just to clarify, is this the cause of the issues for your MCU or suspected?

The library [should ignore the firmware interval](https://github.com/WifWaf/MH-Z19/blob/b0f7fb1065a1fe66cdf3d2cc135c79f1e748166c/src/MHZ19.cpp#L116), although I'm not sure it's working on newer FW versions. To be honest with you, I've been tied down after returning to university for the past year but will have time to implement any changes in June. 

I would recommend giving the [PR_2.0.0 branch](https://github.com/WifWaf/MH-Z19/tree/PR_2.0.0) a try too. It was a newer iteration I was working on. Although If I recall, it's mostly just memory optimisation so might not help (I should add, I would ignore the storage function unless you have a test sensor you don't mind corrupting, I was testing the configuration area).