WifWaf/MH-Z19 #52: Failed communication via UART (ESP32)

edomazzone, 2023-01-15:
Hi everyone,
I'm using ESP32 with MHZ19B co2 sensor. I've changed the BasicUsage sketch to use HardwareSerial 2. After wiring everything the result was:
co2: 0
temperature: -17
[image]

WifWaf, 2023-01-21:
Hey, are you still having this issue?

I think what's happening here is that the CO2 is 0 because the sensor needs calibrating. 

The temperature value is also wrong due to the library as the sensor firmware is often revised, making it awkward to keep compatibility as I only have a few sensors to work with.

Have you tried calibrating the sensor?

edomazzone, 2023-02-20:
Hey, I managed to have data after doing calibration with one sensor that works correctly. Another one, that is from the same order, has results like:
CO2: 5000 ppm
T: 21 °C

The temperature is correct, but the CO2 is fixed at the maximum measure possible even if I've ventilated the room. And changing the top level to 2000 ppm the data are flat at the maximum level even after calibration.
It's possible that I need to make a solder junction in the PCB of the sensor?
I share the data from the correct sensor that after 5 minutes start to send the correct data.
[image]

WifWaf, 2023-02-20:
The library temperature function doesn't always work as it depends on the version of the sensor (the firmware has been changed many times). If you run the [RetrieveDeviceInfo](https://github.com/WifWaf/MH-Z19/tree/master/examples/RetrieveDeviceInfo), I imagine you'll see different versions.

In the case of the CO2 value being stuck at 5000ppm on the second sensor, I'm not sure, calibrating should fix the issue. It's possible the range and/or span were not set correctly by the manufacturer, but I wouldn't suggest calling the functions unless it's a last resort (see [the calibration plus example](https://github.com/WifWaf/MH-Z19/tree/master/examples/CalibrationPlus)).

If you can't return the sensor, then you could try setting the span by blowing it into a bag or something and placing both sensors inside. You could then try and get the CO2 reading on the working sensor to be close to the span/range you want (e.g. 2000 or 5000) and then call the zeroSpan(value) and setRange(value) on the sensor which is stuck at 5000ppm.