WifWaf/MH-Z19 #53: Error Code 3

icsHaLl, 2023-02-14:
Hello 
I use esp32 UART to communication with MH-Z19 MH-Z19C

than, I tried the example ErrorCodes and I got message

Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 129 8 200 3 52 1 ERROR Code: 3
Failed to receive  CO2 value - Error
Response Code: 3
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 129 8 200 3 52 1 ERROR Code: 3
Failed to receive CO2 value - Error
Response Code: 3
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 130 8 198 3 51 1 ERROR Code: 3
Failed to receive  CO2 value - Error
Response Code: 3
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 130 8 198 3 51 1 ERROR Code: 3
Failed to receive CO2 value - Error
Response Code: 3
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 130 8 199 3 50 1 ERROR Code: 3
Failed to receive  CO2 value - Error
Response Code: 3
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 66 77 9 131 8 199 3 49 1 ERROR Code: 3
Failed to receive CO2 value - Error
Response Code: 3

Could you help me please?

Thank you @WifWaf

WifWaf, 2023-02-16:
I'm not sure what's happening here, it might be synchronisation issue.

Have you tried the [RetrieveDeviceInfo.ino](https://github.com/WifWaf/MH-Z19/blob/master/examples/RetrieveDeviceInfo/RetrieveDeviceInfo.ino)?

irrvrsl, 2023-05-11:
Hello! I have similar issue with ESP32 and MH-Z19B. When running [https://github.com/WifWaf/MH-Z19/blob/master/examples/RetrieveDeviceInfo/RetrieveDeviceInfo.ino](RetrieveDeviceInfo.ino) it outputs:
```
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 255 1 133 0 0 0 0 0 122 ERROR Code: 3
Failed to recieve CO2 value - Error
Response Code: 3
```

When running [https://github.com/WifWaf/MH-Z19/blob/master/examples/BasicUsage/BasicUsage.ino](BasicUsage.ino) it always show 0ppm and -17C. I tried to calibrate via hardware method (HD pin to LOW for 7+ sec) and software method (20+ minutes running [https://github.com/WifWaf/MH-Z19/blob/master/examples/Calibration/Calibration.ino](Calibration.ino) on open window. Nothing has changed. I tried to power MH-Z19B with 3.3V and 5V, same results. Any tips would be highly appreciated, thanks!

WifWaf, 2023-05-11:
Try using getCo2(false) and let me know if it works.

irrvrsl, 2023-05-11:
It was totally my mistake, sorry for bothering. For those who came across the same issues: check your RX & TX connections. You should connect them like this:
MH-Z19B **RX** -> *your board* **TX**
MH-Z19B **TX** -> *your board* **RX**

Now sensor outputs:
```
Sent << DEC: 255 1 133 0 0 0 0 0 122  
Received >> DEC: 255 133 6 40 0 0 4 101 228  
CO2 Value successfully Received: 0
Response Code: 1
```

With `getCO2(false)` it outputs:
```
Sent << DEC: 255 1 134 0 0 0 0 0 121  
Received >> DEC: 255 134 1 146 56 0 0 0 175  
CO2 Value successfully Received: 402
Response Code: 1
```
I think that's ok(?), it should calibrate itself within 24 hours

WifWaf, 2023-05-12:
If it's still calibrating, that's fine. It just means that the baseline CO2 value it has stored is too high.

dmtrKovalenko, 2024-05-17:
I am getting something similar. Every second measurement I get the errorCode 3 and then it makes a normal measurement. I suppose it is not a default behavior, right? Maybe you have an idea what should I do? Try to get a new a sensor?

Firmware Version: 05.20
Range: 5000
Background CO2: 428
Temperature Cal: 40
ABC Status: ON