WifWaf/MH-Z19 #54: Cannot connect the sensor properly

vinzenzwolf1, 2023-04-28:
I used an Arduino Micro and just used your example code and connected the sensor like this:

- 5V (Arduino) to Vin (MH-Z19B)
- GND to GND
- Rx (Arduino) to Tx (MH-Z19B)
- Tx (Arduino) to Rx (MH-z19B)

Unfortunately, I got this Error:
[image]

I changed the value of the rx_pin and tx_pin like this:
#define RX_PIN 10
#define TX_PIN 11

When I look at the Sensor, I can see the IR light turning on every 2 seconds or so. On the arduino, the tx light flashes on and of, but the rx light is permanentely of. 

What can I do?

WifWaf, 2023-04-28:
I'm not sure which Arduino Micro board you are using, however some versions operate at 3.3V, so there could be a serial voltage compatibility issue.

I'm not that familiar with the Arduino Micro... however, have you tried using Serial1 and the respective pins instead?

vinzenzwolf1, 2023-04-29:
The Arduino Micro has a 5V Port, I used that one. 
[image]

The operating voltage of my arduino is 3 Volts. 

Sorry, which port should I try?

WifWaf, 2023-04-29:
If the operating voltage is 3V, then the serial port might also be using 3V. Try using the 3.3V to power the MHZ19 instead of the 5V.

Serial1 is a pre-defined serial instance, you can pass it to the library just like you can pass the software serial instance. You'll have to read up on it as I don't know much about the Micro, however, try using the 3.3V to power the MHZ19 first.

vinzenzwolf1, 2023-04-29:
I tried using the 3V port, but unfortunately it wasn't successfull. After some research I found out about the same as you told me (to change to Serial1). As I'm new into microcontroller and in general C++ I need to make some research on Serial ports to understand the general idea. Thanks a lot for the hints! I'll keep you updated!

WifWaf, 2023-05-01:
No problem.

The Serial1 type of definitions are not very self-explanatory and also inconsistent between MCU (as each has its own unique Arduino framework). Best of luck.